nordicsemi/Android-nRF-Toolbox #4: some issue with chip

myselfgaurav, 2015-07-01:
i am using nordic ble but issue it that it get auto disconnected i want to know is there 

1: some sleep time 
2: how can i debug hardware configuration from chip

philips77, 2015-07-07:
Hi, you need to provide more details about your issue. 
What phone are you testing on? What device/chip do you connect with?
What is the issue exactly? When the device gets disconnected?
Is you application designed to disconnect after some time of inactivity? 
What do you mean by saying debug hardware configuration?

myselfgaurav, 2015-07-07:
i am testing it on cyanogenmod 5.0.2 Android. i am using nordic N51822 ble chip.

issue it that it gets disconnected some times say after 1-2 hour automatically.

i have got prebuild chip from some distributer , how can i figure if problem is with android code or chip and also way to debug chip configuration

philips77, 2015-07-07:
Could you find or post your issue on https://devzone.nordicsemi.com/questions/ portal? I guess it's more related to Android or nRF51 itself than the nRF Toolbox app. As far as I remember the phone I'm working on was several times still connected after a night, but I didn't test it very well. What profile are you testing?

closed 2015-07-21 by philips77