nordicsemi/Android-nRF-Toolbox #8: Can not resolve symbol 'DfuProgressListener'

yahch, 2015-08-12:
when I open  'Android-nRF-Toolbox' ,a problem occured

in file DfuActivity.java  

import no.nordicsemi.android.dfu.DfuProgressListener;
import no.nordicsemi.android.dfu.DfuProgressListenerAdapter;
import no.nordicsemi.android.dfu.DfuServiceListenerHelper;
import no.nordicsemi.android.dfu.DfuServiceInitiator;

ERROR: Can not resolve symbol 'DfuProgressListener'
ERROR: Can not resolve symbol 'DfuProgressListenerAdapter'
ERROR: Can not resolve symbol 'DfuServiceListenerHelper'
ERROR: Can not resolve symbol 'DfuServiceInitiator'

but DFU Library has already placed with Android-nRF-Toolbox directory

Settings.gradle:

project(':dfu').projectDir = file('../DFULibrary-master/dfu')

Please, how can I  solves this problem?

philips77, 2015-08-12:
Hi,
You need to set up the project just like shown on [this](https://github.com/NordicSemiconductor/Android-nRF-Beacon) page. This is another project, but the principles are the same. The DFU Library needs to be in DFULibrary folder in the same root folder like shown on the image there.

philips77, 2015-08-12:
If you changed the path in the Settings.gradle, I guess it should compile without problems. I'm not sure where else should you change the paths... Try cleaning the project.

yahch, 2015-08-13:
oh I was too careless,I should clone the project  rather than download the zip file,so the DFU-Library DfuProgressListener.java missing...  and thank you very very very much!   :)

closed 2015-08-13 by yahch