nordicsemi/Android-nRF-Toolbox #11: App not detecting a Heart Rate Service

xcrespo, 2015-09-22:
I have a nRF51822 and I'm using [mbed](https://mbed.org/) platform to integrate a Heart Rate Service, however this app does not detect the service when scanning. I've also implemented the UART Service and it does find it. 

I've found that other people have [a similar problem](https://developer.mbed.org/questions/6969/This-cannot-connect-to-nRF-Toolbox/) because Bluetooth was sending incorrect advertising packets, but this bug was already solved in mbed. I've checked with nRF Master Control Panel and advertising is working properly. With this app I can also detect the Heart Rate Service and read its characteristics.

Is it possible that the BT scanner filter, which is looking for an ```UUID = 0000180D-0000-1000-8000-00805f9b34fb```, doesn't take into account the HR Service which uses only the 16-bit (```UUID = 180D```) (because it is an adopted service)?

philips77, 2015-09-23:
Hi, it should work with the 16-bit UUID in adv packet. I have an example here where the device advertises with Complete list of 16-bit Service UUIDs: 0x180D, 0x180F, 0x180A and is being find in the HRM profile. I can connect to it as well. Could you send the adv packet log from the nRF MCP app? (long click the device and select Share action in the menu).

xcrespo, 2015-09-23:
Hi @philips77, thank you for your help. The advertising packet log is:
```
Name: nRF
Address: F6:38:A9:EB:AC:44
RSSI: -74
Last advertisement packet:
Raw data: 0x02010608084D545374726170
```

philips77, 2015-09-23:
You need to add the 0x180D service to the adv packet in order to make it visible in the scanner dialog in HRM profile. Add the following bytes to your adv packet:
0x02010608084D545374726170**03030D18**
These means:
03 - length: 3 bytes
03 - type: Complete List of 16-bit Service Class UUIDs
0D18 - UUID in the Little Endian (180D = HRS)

xcrespo, 2015-09-23:
I see. And why does it work with the UART service? Its UUID is ```6E400001-B5A3-F393-E0A9-E50E24DCCA9E``` and it is not on the adv packet, isn't it?

philips77, 2015-09-23:
No, in the DFU and UART profiles the scanner does not filter out any devices. These services are not something you should announce, as they are used for development/firmware update. But HR service is a good example of such that helps to filter devices which makes the user experience better.

If you program different applications on the same device you may need to clear the service cache on Android. The easiest way is to connect to such device with nRF MCP and disconnect (if Refresh cache option is set to Always in the settings) or connect and select Refresh services in the menu.

xcrespo, 2015-09-23:
It makes sense. I'll try your solution but for now we can consider this issue closed.
Thank you very much.

closed 2015-09-23 by xcrespo