nordicsemi/Android-nRF-Toolbox #14: Mode DFU

arebgon, 2016-02-26:
Error: "GATT CONN TIMEOUT"

onUploadClicked - Uri content://com.google.android.apps.docs.storage/document/acc%3D3%3Bdoc%3D84 - Path null - InitUri content://com.google.android.apps.docs.storage/document/acc%3D3%3Bdoc%3D84 - InitPath null
02-26 13:27:48.809 11494-16278/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: connect() - device: 3B:90:58:F6:36:37, auto: false
02-26 13:27:48.809 11494-16278/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: registerApp()
02-26 13:27:48.809 11494-16278/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: registerApp() - UUID=477ed59a-1877-4b6e-b7f1-b4ff04ea544f
02-26 13:27:48.859 11494-11508/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: onClientRegistered() - status=0 clientIf=7
02-26 13:27:49.409 11494-11508/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: onClientConnectionState() - status=0 clientIf=7 device=3B:90:58:F6:36:37
02-26 13:27:49.419 11494-11508/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: discoverServices() - device: 3B:90:58:F6:36:37
02-26 13:27:50.669 11494-11508/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: onSearchComplete() = Device=3B:90:58:F6:36:37 Status=0
02-26 13:27:51.709 11494-16278/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: setCharacteristicNotification() - uuid: 00001531-1212-efde-1523-785feabcd123 enable: true
02-26 13:27:57.299 11494-12042/no.nordicsemi.android.nrftoolbox D/BluetoothGatt: onClientConnectionState() - status=8 clientIf=7 device=3B:90:58:F6:36:37
02-26 13:27:57.299 11494-12042/no.nordicsemi.android.nrftoolbox E/DfuBaseService: Connection state change error: 8 newState: 0
02-26 13:27:57.299 11494-16278/no.nordicsemi.android.nrftoolbox E/DfuBaseService: Unable to write Init DFU Parameters (error 32776)

Not working DFU.

philips77, 2016-02-27:
Hi,
Here's what I recommend to test:

1. Does this issue occur every time you try? Looks like the device does not respond. Could you try it with the DFU Bootloader from the SDK? Does your device successfully jump to the bootloader mode, or doesn't need to?
2. I've made some changes in the DFU library in version 1.16.2, could you try update with nRF Toolbox version 1.16.1 (available in Releases in this GitHub project). If it fails on 1.16.2 but works on 1.16.1 I'll have a look at it on Monday morning. I have the Samsung S6. 
3. Could you try updating your device with nRF Master Control Panel 4.0.5 and post the log here, if it fails? They share the same library for DFU.

arebgon, 2016-02-29:
Hi!

Yes, the Bluetooth device is in DFU mode. I try to do what he tells me. 

Thank, you.

ravenet, 2016-03-01:
Dear Creator,

I need to ask one question,but it is an issue so that I do not create an issue. How to read weight scale features value. I checked in your code, I could not find the parser. 

Could you please help me on it? thanks.

philips77, 2016-03-01:
Hi,
Here's code you may find useful. It parses the value and returns it as a String, but based on that you should know how to return value in format you need.

```java
import android.bluetooth.BluetoothGattCharacteristic;

public class WeightScaleFeatureParser {
	private static final int TIMESTAMP_SUPPORTED = 0x01; // 1 bit
	private static final int MULTIPLE_USERS_SUPPORTED = 0x02; // 1 bit
	private static final int BMI_SUPPORTED = 0x04; // 1 bit
	private static final int WEIGHT_MEASUREMENT_RESOLUTION = 0x78; // 4 bits
	private static final int HEIGHT_MEASUREMENT_RESOLUTION = 0x0380; // 3 bits

	/**
	 * Parses the Weight Scale Feature characteristic.
	 * 
	 * @param characteristic the characteristic to parse
	 * @return Weight scale feature in human readable format
	 */
	public static String parse(final BluetoothGattCharacteristic characteristic) {
		final byte[] bytes = characteristic.getValue();
		if (bytes.length != 4) {
			return "Incorrect data length (4 bytes expected)";
		}

		final int flags = characteristic.getIntValue(BluetoothGattCharacteristic.FORMAT_UINT32, 0);

		final boolean timestampSupported = (flags & TIMESTAMP_SUPPORTED) > 0;
		final boolean multipleUsersSupported = (flags & MULTIPLE_USERS_SUPPORTED) > 0;
		final boolean bmiSupported = (flags & BMI_SUPPORTED) > 0;
		final int weightResolution = (flags & WEIGHT_MEASUREMENT_RESOLUTION) >> 3;
		final int heightResolution = (flags & HEIGHT_MEASUREMENT_RESOLUTION) >> 7;

		final StringBuilder builder = new StringBuilder();
		builder.append("Flags:\n");
		if (timestampSupported)
			builder.append("Timestamp Supported\n");
		if (multipleUsersSupported)
			builder.append("Multiple Users Supported\n");
		if (bmiSupported)
			builder.append("BMI Supported\n");
		builder.append("Weight Measurement Resolution: ");
		switch (weightResolution) {
			case 0:
				builder.append("Not Specified\n");
				break;
			case 1:
				builder.append("0.5 kg or 1 lb\n");
				break;
			case 2:
				builder.append("0.2 kg or 0.5 lb\n");
				break;
			case 3:
				builder.append("0.1 kg or 0.2 lb\n");
				break;
			case 4:
				builder.append("0.05 kg or 0.1 lb\n");
				break;
			case 5:
				builder.append("0.02 kg or 0.05 lb\n");
				break;
			case 6:
				builder.append("0.01 kg or 0.02 lb\n");
				break;
			case 7:
				builder.append("0.005 kg or 0.01 lb\n");
				break;
			default:
				builder.append("Reserved for future use\n");
				break;
		}
		builder.append("Height Measurement Resolution: ");
		switch (heightResolution) {
			case 0:
				builder.append("Not Specified\n");
				break;
			case 1:
				builder.append("0.01 meter or 1 inch\n");
				break;
			case 2:
				builder.append("0.005 meter or 0.5 inch\n");
				break;
			case 3:
				builder.append("0.001 meter or 0.1 inch\n");
				break;
			default:
				builder.append("Reserved for future use\n");
				break;
		}
		builder.setLength(builder.length() - 1);
		return builder.toString();
	}
}
```

ravenet, 2016-03-01:
Hi Philip,

Thank you and I will try it and let you know.

ravenet, 2016-03-08:
Dear Philip,

I have used the parser with one of the existing activities code to read value, but it is not return value. I suspect that it could be due to my Activity which i created based on your existing Activities is incorrect.

Could you help me add weight scale features in code as you have experience in this protocol.

I really appreciate it.

philips77, 2016-03-14:
Hi @ravenet, I'm quite busy with my current project right now and tomorrow I go for vacations for two weeks so I won't be able to help you right now. Maybe someone else could help?

ravenet, 2016-03-14:
Dear Philip,
I have done it, and I also willing to share in this project, please let me know how to do it so.

thanks

arebgon, 2016-03-21:
Dear Philips,

I send the my application log for you to check the error.

nRF Master Control Panel, 2016-03-21
PDFU (0F:A8:CF:DB:73:F2)
D 11:02:53.967 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V 11:02:54.501 Connecting to 0F:A8:CF:DB:73:F2...
D 11:02:54.515 gatt = device.connectGatt(autoConnect = false)
D 11:02:54.597 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 11:02:54.620 Connected to 0F:A8:CF:DB:73:F2
D 11:02:54.643 wait(600ms)
V 11:02:55.259 Discovering services...
D 11:02:55.271 gatt.discoverServices()
D 11:02:55.285 [Callback] Services discovered with status: 0
I 11:02:55.296 Services discovered
V 11:02:55.320 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
D 11:03:01.754 [Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
E 11:03:01.796 Error 8 (0x8): GATT CONN TIMEOUT
D 11:03:01.806 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
I 11:03:01.814 Disconnected
D 11:03:01.844 gatt.close()

thanks

arebgon, 2016-03-21:
Dear Philips,

Sorry, this is the log. 

nRF Master Control Panel, 2016-03-21
PDFU (27:16:18:BF:39:00)
D 11:13:53.993 [Broadcast] Action received: android.bluetooth.device.action.ACL_CONNECTED
V 11:13:54.908 Connecting to 27:16:18:BF:39:00...
D 11:13:54.922 gatt = device.connectGatt(autoConnect = false)
D 11:13:54.984 [Callback] Connection state changed with status: 0 and new state: CONNECTED (2)
I 11:13:54.994 Connected to 27:16:18:BF:39:00
D 11:13:55.009 wait(600ms)
V 11:13:55.620 Discovering services...
D 11:13:55.638 gatt.discoverServices()
D 11:13:55.659 [Callback] Services discovered with status: 0
I 11:13:55.672 Services discovered
V 11:13:55.697 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Appearance [R] (0x2A01)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Generic Attribute (0x1801)
Device Firmware Update Service (00001530-1212-efde-1523-785feabcd123)
- DFU Packet [WNR] (00001532-1212-efde-1523-785feabcd123)
- DFU Control Point [N W] (00001531-1212-efde-1523-785feabcd123)
   Client Characteristic Configuration (0x2902)
D 11:14:01.801 [Callback] Connection state changed with status: 8 and new state: DISCONNECTED (0)
D 11:14:01.832 [Broadcast] Action received: android.bluetooth.device.action.ACL_DISCONNECTED
E 11:14:01.850 Error 8 (0x8): GATT CONN TIMEOUT
D 11:14:01.870 gatt.close()

thanks

ravenet, 2016-03-22:
Hi Arebgon,

Can you give us more details when you are getting this error and what are you trying to do ?
Thanks

closed 2016-10-19 by philips77