nordicsemi/Android-nRF-Toolbox #15: multiple devices, have what good way

amibition521, 2016-04-01:
I want to connect to multiple devices, have what good way

philips77, 2016-04-14:
Hi,
If you want to connect to multiple devices I recommend having one instance of BluetoothGattCallback per device. In nRF MCP I have a class Connection, that implement this interface, has BluetoothGatt reference and API to send and read data. Then, in the service, I keep a Map of connected devices.
Right now there is no such example in the nRF Toolbox app, but it is possible to connect to more than one device, but they must be connected from different profiles. Also, not all of them support them. But e.g. RSC and CSC or Proximity can (those with Service).

amibition521, 2016-04-28:
I understand your ideas, I must for each connected to the equipment to create a BluetoothGattCallback.

amibition521, 2016-04-28:
now I  face a situation on the following: the premise is that I will be connected with the equipment that are stored locally like macaddress, when I off the device, I start to do automatic connection device, but now the callback connection for a long time, about 35s, any good solution can shorten the feedback time.

philips77, 2016-09-30:
Hi, the timeout supervision on Android is set to 20 sec on any Connection Priority. You may request smaller from the device though. 
I will start working on adding multiple connections to Proximity profile soon and hopefully manage to finish it before we start a new project.

philips77, 2016-10-19:
Hi, I've just published nRF Toolbox 2.0 which supports now multiple connections in a profile (Proximity).
Additionally there has been done some tweaks in BleManager and its callback to support it.
Hope that helps.

I'm closing the issue. In case you found something not working please open another one.

closed 2016-10-19 by philips77