nordicsemi/Android-nRF-Toolbox #17: Custom `BleManager` implementation

metc, 2016-05-16:
Hi Nordic team,

*This is not really an issue but more a discussion.*
I am an embedded BLE and Android developer. I saw that you provide the [Android-Scanner-Compat-Library](https://github.com/NordicSemiconductor/Android-Scanner-Compat-Library) project as a library which is great.
I also see that you made a custom `BleManager` implementation to manage the low level communication with Bluetooth Smart devices. This basically add a queue of requests (ATT commands) which is no available in the Android BLE stack.

I also made my own implementation of the `BleManager` class which also includes a BLE request queue, timeout, support read/write request retries, etc. I saw that your `BleManager` implementation is available in many places ([see this search on Github](https://github.com/search?utf8=%E2%9C%93&q=org%3ANordicSemiconductor+in%3Apath+%22BleManager.java%22&type=Code&ref=searchresults)):

1. [NordicSemiconductor/Android-Digital-Bird](https://github.com/NordicSemiconductor/Android-Digital-Bird/blob/b7eeca1c1e8f4c181bc8769b8a749bc3e41949d5/app/src/main/java/no/nordicsemi/android/digitalbird/ble/BleManager.java)
2. [NordicSemiconductor/Android-nRF-Blinky](https://github.com/NordicSemiconductor/Android-nRF-Blinky/blob/ec4c67fbdd486dab6c47cfc1010a70d79ba3d7df/app/src/main/java/no/nordicsemi/android/blinky/profile/BleManager.java)
3. [NordicSemiconductor/Android-nRF-Toolbox](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/b4fdad1ebc5517e6fadc7181caa6a301815e9f48/app/src/main/java/no/nordicsemi/android/nrftoolbox/profile/BleManager.java)
4. [NordicSemiconductor/Android-nRF-Toolbox Wear](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/b4fdad1ebc5517e6fadc7181caa6a301815e9f48/wear/src/main/java/no/nordicsemi/android/nrftoolbox/ble/BleManager.java)

All these versions seems to be a bit different in each project. @philips77 it seems you are the developer of this. Which version is the most recent ? Do you plan (Nordic) to provide the BLE classes as a library ? Thanks in advance.

philips77, 2016-05-16:
Hi, the most recent one (constantly supported) is in nRF Toolbox (and Wear for Android Wear). I guess the differences aren't big... if any? Others may be a forked from this one from some moment in time.

metc, 2016-05-17:
Hi, thanks for your reply. The changes are minor yes, but they are some because the implementation is  not totally projet independent.

I would like to known how you handle the fact that the BLE stack can be busy. From the `nextRequest` method, you call the low-level primitives to write to a characteristic (ATT write command for instance), but you don't handle the return code given by the stack...  How do you handle messages retransmissions ? What is the common pattern do you use ?

philips77, 2016-05-17:
We don't handle retransmissions. Actually, I don't know if it would be that easy... from my experience, when Bluetooth hangs, you have to reset the adapter, or sometimes the phone as it just automatically turns on again (so in fact can't even turn off). It's such a critical problem, that users see it as "this application doesn't work and killed my Bluetooth! 0 stars!".

metc, 2016-05-17:
Thanks for answer. I will investigate a bit more but if I remember correctly any ATT read/write commands from the `BluetoothGattCharacteristic` class can return `false`, and this was the case quite frequently. The reason why `false` is returned is not very clear for me (radio busy or something similar) and don't seems to be documented. If `false` is returned, IMO it should be handle correctly, but I did not find a good pattern yet.

closed 2016-05-23 by metc