nordicsemi/Android-nRF-Toolbox #22: how to unregister the intentreceiver when dfu complete

TedWithAHammer, 2016-09-26:
[image]
what i miss ,i just need access dfu module

philips77, 2016-09-27:
This has been fixed recently. Please update your DFU library to the newest version, 1.0.4.

TedWithAHammer, 2016-09-27:
so what is the problem ?which class?

philips77, 2016-09-27:
Follow those 2 commits
https://github.com/NordicSemiconductor/Android-DFU-Library/commit/0ce71c3f59b8adc2d287a55cca328311fb743328 
https://github.com/NordicSemiconductor/Android-DFU-Library/commit/c1455e2b6fe9debf0eba61dbe18e3d38d83211ac
Which version of nRF Toolbox are you using?

TedWithAHammer, 2016-09-28:
1.0.0

TedWithAHammer, 2016-09-28:
thank you ,i've downloaded 1.0.0 ,i'll check the differences

closed 2016-09-30 by philips77