nordicsemi/Android-nRF-Toolbox #23: HRM can not find devices

Kevin-Leung, 2016-09-30:
Hi,I found HRM function module can not search any BLE devices, Did I miss something?

philips77, 2016-09-30:
Hi, thanks for the report. I'll check it right away and come back to you soon.

philips77, 2016-09-30:
Are you sure your device advertises with the HRM UUID (0x180D)? The HRM scans with the UUID filter. You may check if your device advertises using either UART profile in nRF Toolbox (there is no filter there), or using nRF Connect application.

Kevin-Leung, 2016-10-02:
HI, It was the filter problem,I removed the filter and then it works well.thanks for you attention.

closed 2016-10-02 by Kevin-Leung

swapnilghadge2390, 2018-10-19:
@Kevin-Leung Can you please guide me which filter you have removed as I am facing the same issue. Thanks in advance