nordicsemi/Android-nRF-Toolbox #28: Compatibility with chromebook

lilliesAndRoses, 2017-03-16:
Hi,
I tried using this app (UART) on Chromebook, since android apps through PlayStore are now supported there. I could not discover any services through this app and hence could not communicate with the NRF DK board. 
Surprisingly, the NRF-Connect app did not have this problem. NRF-Connect was also taken from PlayStore. But it could find services and could communicate with the DK board. 
Why does the "toolbox" app not work while "connect" app does?

philips77, 2017-03-16:
To be honest: no idea. Could you try dfu profile? 
Also, did you granted location permission to the toolbox app? Could you post a screenshot of nrf connect on Chromebook? Unfortunately we don't have any here and I'm curious how does it look like on such big screen.

lilliesAndRoses, 2017-03-16:
We have granted all the permissions required. DFU profile also does not work. It fails finding the DFU service itself. 
The Android app we have written also does not get any service handles when run on Chromebook.  While checking for other apps, we used NRF-toolbox and NRF-connect as references. Out of these, NRF-connect worked.
Is there a difference in the way these two apps connect to devices or discover services?

Vhaiyue, 2018-01-02:
Same as you do，when I send zip  to BLE,some phone could success，But there are always some equipments couldnt,such as huawei4x.

roshanrajaratnam, 2018-01-02:
@Vhaiyue Could you create a new issue for your issue please? It will help us to track and fix issues much faster. Thanks

philips77, 2019-01-30:
Hi, sorry for the super long delay. Did you manage to finally run the app on Chromebook? we still on't have that device to test, but as Chromebooks get on popularity we could try getting one.

I can't think of any reason why nRF Connect would work and nRF Toolbox wouldn't.

franckgy, 2026-02-12:
Hello, this problem is still present, with ChromeOs v145.0.7632.38 and latest version of Nrftoolbox and Nrfconnect