nordicsemi/Android-nRF-Toolbox #31: Can not connect to devices with SDK 13 from android 4.4.4

lilliesAndRoses, 2017-05-11:
Hi,
I use the NRF-toolbox app's UART feature for connecting to nordic devices.
Android version is 4.4.4 and the device used is a Samsung device. 
Everything was working fine when nordic device was using SDK 11, Softdevice S132 v3.
But after migration to SDK 13 and softdevice S132 v4.0.2, connection does not go through. 
The connect request succeeds, but service discovery fails with GAT_INTERNAL_ERROR (0x81).
This is followed by a disconnection of the device (GATT_ERROR 0x85).
Same behaviour is seen through NRF-connect as well.

Is support for 4.4.4 android dropped from SDK 13, softdevice S132v4.0.2? Or are there any known issues?

Thanks

lilliesAndRoses, 2017-05-16:
Has anyone else seen this issue?

philips77, 2017-05-16:
Hi, yes. I've seen issues with connecting to devices built with SDK 13. Pixel phone works, Nexus 6P doesn't. I still don't know what's the root cause. In SDK 13 support for DLE (Data Length Extension) was added, which is supported by Pixel, but not the Nexus family, but if it's not supported they should just return Not Supported and everything should be like in the old times... I suspect DLE, but it may something very different. Please, ask on DevZone. The support team should handle this issue.

Vhaiyue, 2017-12-29:
I send zip,when progress 10%，the phone get disconessed to ble,but my phone xiaomi4s can 100%，but the huawei4X cant,and oppo r9s get133 state. why the different phone will get the diffrent results.I am crazy with the primission.hope help me

philips77, 2017-12-29:
Hi, could you create another issue, as it's not related to this one.

closed 2019-01-30 by philips77