nordicsemi/Android-nRF-Toolbox #33: Notifications enabled using Write Command

philips77, 2017-06-30:
Prior to Android 6 there was a bug on Android. The `gatt.writeDescriptor(descriptor)` method was using parent characteristic's write type, instead of always using WRITE_DEFAULT (Write Request).
nRF Toolbox was ensuring the write type for normal descriptor writes ([see here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/fe75efa4cd78162f4a769e2d8076870d81c09041/app/src/main/java/no/nordicsemi/android/nrftoolbox/profile/BleManager.java#L644)), but this workaround wasn't applied during enabling notifications/indications ([here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/fe75efa4cd78162f4a769e2d8076870d81c09041/app/src/main/java/no/nordicsemi/android/nrftoolbox/profile/BleManager.java#L729), [here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/fe75efa4cd78162f4a769e2d8076870d81c09041/app/src/main/java/no/nordicsemi/android/nrftoolbox/profile/BleManager.java#L519) and [here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/fe75efa4cd78162f4a769e2d8076870d81c09041/app/src/main/java/no/nordicsemi/android/nrftoolbox/profile/BleManager.java#L550)).
In all those cases it also should ensure correct write type.

closed 2017-06-30 by philips77

samir-eljazovic-sn, 2017-07-03:
Hi @philips77 do you know if there is a reference for this issue on Google Issue Tracker?

samiraga, 2017-07-03:
Hi @philips77 do you know if there is a reference for this issue on Google Issue Tracker?

philips77, 2017-07-04:
https://issuetracker.google.com/issues/37022798 how about this?

samiraga, 2017-07-05:
Thanks @philips77

commit 5f436303 (2017-06-30): Bug fixed: #33