nordicsemi/Android-nRF-Toolbox #36: DFU get DUF FILE ERROR on Google Play version

yhchens, 2017-11-07:
I build it from source code try to figure what happen, but it is fine with debug build. I think google play version has something wrong.

philips77, 2017-11-07:
You are correct. Sorry for that. I've added proguard in this version, this may be the reason. I'll prepare a fix tomorrow.

philips77, 2017-11-07:
DfuBaseService: java.io.IOException: The ZIP file must contain an Application, a Soft Device and/or a Bootloader.

philips77, 2017-11-07:
The error comes from here: https://github.com/NordicSemiconductor/Android-DFU-Library/blob/release/dfu/src/main/java/no/nordicsemi/android/dfu/internal/ArchiveInputStream.java#L245
For unknown reason the archive parser does not find the manifest file.

philips77, 2017-11-07:
Fixed in 2.5.1. Proguard removed manifest classes, so the Manifest object failed to be parsed and that's why it was failing.

philips77, 2017-11-07:
Commit: 29af8d68266302bd84e1a521b1fc3fb30f0a5297

closed 2017-11-07 by philips77