nordicsemi/Android-nRF-Toolbox #38: How to use BleManager in my application?

Wisawing, 2017-11-28:
Hi, I am a beginner developer in Android app so before I start I want to state that it is highly likely that the problem I am encountering is due to my lack of knowledge on application development and not this project itself.

I am trying to use BleManager in my own application to connect to nRf51-DK. So far I have accomplished in doing :
1. Compile, install and run this project on Android.
2. The self compiled project can be used and correctly connected to nRF51-DK

From what I understand, I should only use the BleManager part and it's dependency( Which should exclude the interfaces/logic/control of nRFToolbox itself ). So for starters in order to not mix up my own code with this project I tried importing the whole "app" module of this project into my own project. I got these errors.

Unable to resolve dependency for ':myproject@debug/compileClasspath': Could not resolve project :app.

 >Could not resolve project :app.
Required by:
    project :myproject
 Unable to find a matching configuration of project :app:
     - Configuration 'debugApiElements':
         - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
         - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
         - Found com.android.build.gradle.internal.dependency.VariantAttr 'debug' but wasn't required.
         - Required org.gradle.api.attributes.Usage 'java-api' and found compatible value 'java-api'.
     - Configuration 'debugMetadataElements':
         - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
         - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Metadata'.
         - Found com.android.build.gradle.internal.dependency.VariantAttr 'debug' but wasn't required.
         - Required org.gradle.api.attributes.Usage 'java-api' but no value provided.
     - Configuration 'debugRuntimeElements':
         - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
         - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
         - Found com.android.build.gradle.internal.dependency.VariantAttr 'debug' but wasn't required.
         - Required org.gradle.api.attributes.Usage 'java-api' and found incompatible value 'java-runtime'.
     - Configuration 'releaseApiElements':
         - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found incompatible value 'release'.
         - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
         - Found com.android.build.gradle.internal.dependency.VariantAttr 'release' but wasn't required.
         - Required org.gradle.api.attributes.Usage 'java-api' and found compatible value 'java-api'.
     - Configuration 'releaseMetadataElements':
         - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found incompatible value 'release'.
         - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Metadata'.
         - Found com.android.build.gradle.internal.dependency.VariantAttr 'release' but wasn't required.
         - Required org.gradle.api.attributes.Usage 'java-api' but no value provided.
     - Configuration 'releaseRuntimeElements':
         - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found incompatible value 'release'.
         - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
         - Found com.android.build.gradle.internal.dependency.VariantAttr 'release' but wasn't required.
         - Required org.gradle.api.attributes.Usage 'java-api' and found incompatible value 'java-runtime'.

After try googling and reading about Android studio dependency settings I still could not resolve the issue so then I tried creating a new module in this project instead. But as soon as I add "app" module as a dependency to my own module, the same errors occur.

Would the correct way to cleanly use BleManager be copying part of the source code of this project? Any help or pointer is appreciated. Thank you.

philips77, 2017-11-28:
Hi, just copy the BleManager, BleManagerCallback, and some other classes and interfaces that you'll find required. If you don't need/want the logger you may use normal Log instead of DebugLogger (?) or remove all lines with Logger class. Logger is used to log to the nRF Logger app while your app is running, if such app is installed.

However, you are right, this should be easier. I'll try to move it to a module when find some time.

Wisawing, 2017-11-30:
Thank you for the reply. I have successfully copied over the BleManager by manually selecting only part that it is needed.

The error I posted was actually because I tried to make an application module depended on another application module. After I changed BleManager module to feature module it worked out.

To change module type I have to change the following line in build.gradle

From :
>apply plugin: 'com.android.application'

To :
>apply plugin: 'com.android.feature'

closed 2017-11-30 by Wisawing