nordicsemi/Android-nRF-Toolbox #40: build fail: Could not determine the dependencies of task ':app:compileReleaseJavaWithJavac'.

whyameye, 2018-01-02:
I am building with Gradle on OSX. When executing `gradle build` I get:

```Could not determine the dependencies of task ':app:compileReleaseJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:releaseCompileClasspath'.
   > Could not resolve project :ble.
     Required by:
         project :app
      > Unable to find a matching configuration of project :ble: None of the consumable configurations have attributes.```

philips77, 2018-01-04:
Sorry, you have to check out this: https://github.com/NordicSemiconductor/Android-Ble-library project into the same error folder (i.e. AndroidstudioProjects). If you choose another folder name or destination, edit the settings.gradle file.

syslogic, 2018-03-23:
@philips77 you could publish the library to Maven Central and add it as dependency... have seen a few issues because of not having checked out the other one repository, it seems to be the common pitfall. this would basically make it available in just any project, by adding a single line.
https://medium.com/@nmauti/sign-and-publish-on-maven-central-a-project-with-the-new-maven-publish-gradle-plugin-22a72a4bfd4b
https://docs.gradle.org/current/userguide/publishing_maven.html

philips77, 2018-03-23:
Yes, we have done it with the dfu lib. BLE library in still under development, we will definitely publish it to jcenter/maven when done.
I'll on vacations now, I'll continue to work on it after Easter.

Honey1994, 2018-09-05:
Error:Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not resolve project :chat.
     Required by:
         project :app
      > Unable to find a matching configuration of project :chat: None of the consumable configurations have attributes.

philips77, 2019-01-30:
The BLE Library and BLE Common Library is now on jcenter and Maven Central.

closed 2019-01-30 by philips77

jitheshkramesh, 2019-04-04:
I am getting an error like this.
Please help.

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:preDebugBuild'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not resolve com.google.android.gms:play-services-base:[15.0.1,16.0.0).
     Required by:
         project :app > com.google.android.gms:play-services-location:15.0.1
      > Failed to list versions for com.google.android.gms:play-services-base.
         > Unable to load Maven meta-data from https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-base/maven-metadata.xml.
            > Could not HEAD 'https://dl.google.com/dl/android/maven2/com/google/android/gms/play-services-base/maven-metadata.xml'.
               > Read timed out
   > Could not resolve com.google.android.gms:play-services-basement:[15.0.1,16.0.0).
     Required by:
         project :app > com.google.android.gms:play-services-location:15.0.1
         project :app > com.google.android.gms:play-services-places-placereport:15.0.1
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error
   > Could not resolve com.google.android.gms:play-services-tasks:[15.0.1,16.0.0).
     Required by:
         project :app > com.google.android.gms:play-services-location:15.0.1
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error
      > Skipped due to earlier error

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 50s
Running Gradle task 'assembleDebug'...
Running Gradle task 'assembleDebug'... Done                        51.4s
Gradle task assembleDebug failed with exit code 1

philips77, 2019-04-04:
Did you try this: https://stackoverflow.com/questions/50791632/could-not-find-any-version-that-matches-com-google-android-gmsplay-services-bas

krishnarajan-acumenTec, 2021-07-19:
Could not determine the dependencies of task ':app:compileAssembleReleaseJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:assembleReleaseCompileClasspath'.
   > Could not resolve project :MyModelLibrary.
     Required by:
         project :app
      > The consumer was configured to find an API of a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'assembleRelease', attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm'. However we cannot choose between the following variants of project :MyModelLibrary:
          - debugAndroidTestCompile
          - debugAndroidTestRuntime
          - debugRuntime
          - debugUnitTestCompile
          - debugUnitTestRuntime
          - releaseRuntime
          - releaseUnitTestCompile
          - releaseUnitTestRuntime
        All of them match the consumer attributes:
          - Variant 'debugAndroidTestCompile' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
                  - Provides attribute 'org.jetbrains.kotlin.localToProject' with value 'local to :MyModelLibrary' but the consumer didn't ask for it
          - Variant 'debugAndroidTestRuntime' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
                  - Provides attribute 'org.jetbrains.kotlin.localToProject' with value 'local to :MyModelLibrary' but the consumer didn't ask for it
          - Variant 'debugRuntime' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
                  - Provides attribute 'org.jetbrains.kotlin.localToProject' with value 'local to :MyModelLibrary' but the consumer didn't ask for it
          - Variant 'debugUnitTestCompile' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
                  - Provides attribute 'org.jetbrains.kotlin.localToProject' with value 'local to :MyModelLibrary' but the consumer didn't ask for it
          - Variant 'debugUnitTestRuntime' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
                  - Provides attribute 'org.jetbrains.kotlin.localToProject' with value 'local to :MyModelLibrary' but the consumer didn't ask for it
          - Variant 'releaseRuntime' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
                  - Provides attribute 'org.jetbrains.kotlin.localToProject' with value 'local to :MyModelLibrary' but the consumer didn't ask for it
          - Variant 'releaseUnitTestCompile' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
                  - Provides attribute 'org.jetbrains.kotlin.localToProject' with value 'local to :MyModelLibrary' but the consumer didn't ask for it
          - Variant 'releaseUnitTestRuntime' capability WelcomeT1Management:MyModelLibrary:unspecified declares a component, as well as attribute 'org.jetbrains.kotlin.platform.type' with value 'androidJvm':
              - Unmatched attributes:
                  - Doesn't say anything about com.android.build.api.attributes.BuildTypeAttr (required 'assembleRelease')
                  - Doesn't say anything about its usage (required an API)
[... cut ...]
              - Incompatible because this component declares a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'release' and the consumer needed a component, as well as attribute 'com.android.build.api.attributes.BuildTypeAttr' with value 'assembleRelease'

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 2s