nordicsemi/Android-nRF-Toolbox #41: Module Not Defined

VinayTShetty, 2018-01-02:
After importing the zip file,the android studio is showing. Module not defined.

roshanrajaratnam, 2018-01-02:
Hi, please clone the [Android-BLE-Library](https://github.com/NordicSemiconductor/Android-BLE-Library) repository within the Android-nRF-Toolbox project. The BLE side of things have been moved in to a separate module to make it easier for maintenance. Let us know how it goes.

VinayTShetty, 2018-01-02:
[image]
[image]

## Hi roshan i have cloned and compiled the file as below you told but its showing different error now.

philips77, 2018-01-02:
The Android-BLE-Library project must be cloned into yours DFUupdate folder, and named Android-BLE-Library, without -master.

philips77, 2018-01-02:
Btw, it's not a good idea to name folders with a name of a branch (master), as you may check you and branch in your local repository.

VinayTShetty, 2018-01-02:
Can u please send the single  zip file which has completed all the above process.thanks in advance.

roshanrajaratnam, 2018-01-02:
@VinayTShetty All you have to do is to just move the extracted Android-Ble-Library folder to "Example 1" folder. This Android-Ble-Library module should be located outside our project folder (This case your project folder seems to be Android-nRF-Toolbox-master). Please try this step.

VinayTShetty, 2018-01-02:
Can u please send the single zip file which has completed all the above process as its again showing the same error .thanks in advance.
@roshanrajaratnam @philips77

VinayTShetty, 2018-01-02:
@roshanrajaratnam 
I have extracted it outside as u suggested,but its still the same issue.
[image]
[image]
[image]

roshanrajaratnam, 2018-01-03:
@VinayTShetty check inside the extracted folder to see if its inside another directory of the same name. I think when you extracted, it went in to a sub folder within the folder. Hopefully this should fix your issue and remember to remove "-master". If you look at the settings.gradle file in the project you should see how the folder should look like.

VinayTShetty, 2018-01-03:
@roshanrajaratnam 
Its seriously not helping its getting more confused.I request you to send a single zip folder which has completed all the above process. so that i can compile and run.

roshanrajaratnam, 2018-01-03:
Here you go, it's not that hard. Just open the toolbox project from Android studio and you are good to go.
[example.zip](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/files/1600177/example.zip)

roshanrajaratnam, 2018-01-04:
@VinayTShetty is your issue resolved now?

philips77, 2018-01-04:
Duplicate of #40

VinayTShetty, 2018-01-04:
@roshanrajaratnam 
Thank you so much for your effort.
 I got to know how to do it.
Am really thank full to you for Sending that Zip file.
But when i open in android studio application its showing Missing Feature Watch.
How to disable that watch feature.
[image]
[image]

philips77, 2018-01-04:
On the right-top corner of AS, next to the green Play icon, there is a drop-down. Select 'app', not 'wear'.

VinayTShetty, 2018-01-05:
@roshanrajaratnam 
@philips77 
Thank you Both for Guiding me till here. 
Its Working like a charm,
But i have some question below please answer it..

**Question 1:-**When i click the select Device Button(Screen Shot 2),the scan is not showing my device.

**Question 2:-**I need only DFU firware update to be implemented to my XYZ application,
              i don t require rest of the things which they have implemented like BPM,UART,...etc.
              Can u please guide me on How to get only DFU firmware update code both java and XML and Manifest.

**Question 3:-**My main requirement is devloping an application, that updated DFU through cloud.Not by selecting the Zip files.
           Is that possible..?
           Where i can find any reference for cloud update..?
           Can u both guide me on Update of DFU through Cloud.
[image]

roshanrajaratnam, 2018-01-05:
@VinayTShetty it would be great if you can post these questions on https://devzone.nordicsemi.com/questions/ as the issues created here are solely for supporting issues related to the app it self. We can take it ahead from there.

closed 2018-01-05 by philips77

VinayTShetty, 2018-01-06:
@roshanrajaratnam 
@philips77 
Can u help me on this issue please link have posted below
https://github.com/NordicSemiconductor/Android-nRF-Connect/issues/53

philips77, 2018-01-06:
No. Please don't link to unrelated issues on GitHub just to get out attention.