nordicsemi/Android-nRF-Toolbox #43: can't choose hex or bin file for OTA

zassen, 2018-01-16:
when I use DFU to flash my Application, I can't find hex or bin file in file manager App.
My phone android version is 8.0

philips77, 2018-01-16:
If you are using native file chooser click the menu in top-right corner and enable *Show internal storage*. Then open the left menu and select your phone's name to get into local directories.
If it's not your case, please use a different app, check if your files are there, etc.
What you described is not an issue of this app so I'm closing it.

closed 2018-01-16 by philips77

zassen, 2018-01-16:
what you described is not my case. I check the file is available. just can't show in app. I recompile the apk after midify the mime type to application/hex. the hex can show in my app.  my file manager app is native app on xiaomi 6

philips77, 2018-01-16:
Well... In my opinion application/hex is (unfortunately) not a valid mime type. There is no defined one, so we rather use application/octet-stream, which works on other file browsers. Unfortunately I can't predict what app are you using, and I can't set more than one data type...
I'd rather say that the native for browser on xiaomi should show hex when this type is requested to be compliant with other browsers.

zassen, 2018-01-17:
maybe can try "application/*" for mime type