nordicsemi/Android-nRF-Toolbox #44: about bonding information is not

ArthurHCL, 2018-03-22:
At the beginning, the device is bound to the app and runs perfectly; after that, the device reflashes the program, the phone clears the device binding, but the device binding information in the app remains unchanged, causing the device to always fail to match due to binding information. And not connected. How to remove the binding information in the app?

philips77, 2018-03-22:
In DFU settings switch Keep bond to ON. It is by default on in nRF Connect, as dar as I remember.

ArthurHCL, 2018-03-23:
No use, I click to delete the binding information, but can not succeed. Excuse me, what binding information is stored in the file on the mobile phone, can you give a path, do I manually delete it?

philips77, 2019-01-30:
Hi @ArthurHCL, did you solve your problem?

The bond information is stored by the system and there is no access to it. You may only remove the bond and bond again. Also, similar bond information is stored on the remote bonded device. Removing bond on Android doesn't mean that the bond is also removed from the device. There is a [Bond Management Service](https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.bond_management.xml) that can be used to remove bond on a remote device, but it has to be implemented there. nRF Toolbox and DFU library don't use this service.

During DFU the bond manager memory on your device may be erased or moved to different location (when SoftDevice is updated). This will effectively remove the bond information from the device. But if your update does not affect this memory page you may set Keep Bond switch in nRF Toolbox and nRF Connect to ON, then the app will not remove the bond information from the phone, so reconnection will be possible after the firmware was updated.