nordicsemi/Android-nRF-Toolbox #45: HTS doesn't run on HTC10 Evo (Android 7)

DuyTrandeLion, 2018-03-28:
Hello, I am using the app on my HTC10 Evo which is using Android 7.0
All the services run well except the HTS. The device is recognized but when I connect, the values are displayed. I test the iOS nRF Toolbox and it recognized my device and received the temperature value. 
Anything I can do to fix this? Thank you very much

philips77, 2018-03-28:
As far as I remember in HTS you have to click a button on the DK (which?) to send an indication with a new temp value. Did you try this?

DuyTrandeLion, 2018-03-28:
Thank you for your reply,
I use some different DKs (also with mbed support), there is no button mentioned in the example. Beside, the temperature value changes almost every 2 seconds.

philips77, 2019-01-30:
Hi @DuyTrandeLion, did you manage to solve your problem? This app has been a bit forgotten and I'm trying to dig it up.

DuyTrandeLion, 2019-01-30:
Hi philips77,
I got my problem solved. Thank you for your support. The app runs well but sometimes it crashes on Android for no reason.

philips77, 2019-01-30:
It's good there is no reason for the crash! Otherwise we would have to fix something! ;)
Closing the issue then.

closed 2019-01-30 by philips77