nordicsemi/Android-nRF-Toolbox #49: Unable to find a matching configuration of project :wear:

lirunqiuqiu, 2018-06-04:
Android version is 3.12,  gradle version is 4.1

Unable to find a matching configuration of project :wear:
  - Configuration 'debugApiElements':
      - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
      - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
      - Found com.android.build.gradle.internal.dependency.VariantAttr 'debug' but wasn't required.
      - Required org.gradle.api.attributes.Usage 'java-api' and found compatible value 'java-api'.
  - Configuration 'debugMetadataElements':
      - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
      - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Metadata'.
      - Found com.android.build.gradle.internal.dependency.VariantAttr 'debug' but wasn't required.
      - Required org.gradle.api.attributes.Usage 'java-api' but no value provided.
  - Configuration 'debugRuntimeElements':
      - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
      - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
      - Found com.android.build.gradle.internal.dependency.VariantAttr 'debug' but wasn't required.
      - Required org.gradle.api.attributes.Usage 'java-api' and found incompatible value 'java-runtime'.
  - Configuration 'releaseApiElements':
      - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found incompatible value 'release'.
      - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
      - Found com.android.build.gradle.internal.dependency.VariantAttr 'release' but wasn't required.
      - Required org.gradle.api.attributes.Usage 'java-api' and found compatible value 'java-api'.
  - Configuration 'releaseMetadataElements':
      - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found incompatible value 'release'.
      - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Metadata'.
      - Found com.android.build.gradle.internal.dependency.VariantAttr 'release' but wasn't required.
      - Required org.gradle.api.attributes.Usage 'java-api' but no value provided.
  - Configuration 'releaseRuntimeElements':
      - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found incompatible value 'release'.
      - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found incompatible value 'Apk'.
      - Found com.android.build.gradle.internal.dependency.VariantAttr 'release' but wasn't required.
      - Required org.gradle.api.attributes.Usage 'java-api' and found incompatible value 'java-runtime'.

philips77, 2018-06-04:
Switch from wear to app in the drop down menu next to the Run button in Android Studio.

philips77, 2018-06-04:
It's that you don't have a wear device to run on. Just change to app and it will run on the phone.

lirunqiuqiu, 2018-06-04:
Thanks for your suggestion. Didn't the app should depend on wear ? I can't run the app. Request class is in wear.
[image]

philips77, 2018-06-19:
This should be fixed now.

closed 2018-06-19 by philips77