nordicsemi/Android-nRF-Toolbox #51: Unable to find writeCharacteristic(BluetoothGattCharacteristic) in Android-nRF-Toolbox

R3MX13, 2018-06-07:
In class CGMSManager,  GlucoseManager, ProximityManager.

error: no suitable method found for writeCharacteristic(BluetoothGattCharacteristic)
method BleManager.writeCharacteristic(BluetoothGattCharacteristic,byte[]) is not applicable
(actual and formal argument lists differ in length)
method BleManager.writeCharacteristic(BluetoothGattCharacteristic,byte[],int) is not applicable
(actual and formal argument lists differ in length)

In Ble Manager I add the missing function

	/**
	 * Writes the characteristic value to the given characteristic.
	 * The write type will be read from the characteristic.
	 *
	 * @param characteristic the characteristic to write to
	 * @return true if request has been enqueued
	 */
	protected final boolean writeCharacteristic(final BluetoothGattCharacteristic characteristic) {
		return enqueue(Request.newWriteRequest(characteristic, characteristic.getValue()));
	}

philips77, 2018-06-07:
True, modify it to use a variant with byte[] instead.

philips77, 2018-06-07:
I've updated the lib yesterday and didn't have time to update the project.

closed 2018-06-19 by philips77