nordicsemi/Android-nRF-Toolbox #52: getGattCallback() in GlucoseManager errors "Method does not override method from its superclass" (Problems with GlucoseManager.class)

Abushawish, 2018-06-08:
Hello, first I wanted to thank you for this amazing library, however I would appreciate some help. I'm not sure if I'm misunderstanding, but I'm running into many problem within the GlucoseManager class.

Within the GlucoseManager which extends BleManager<GlucoseManagerCallbacks>:

```
public class GlucoseManager extends BleManager<GlucoseManagerCallbacks> {...{
```

There is a method which overrides getGattCallback():

```
@Override
protected BleManagerGattCallback getGattCallback() {
  return mGattCallback;
}
```

However, Android Studio shows an error:

```
 "Method does not override method from its superclass"
```

But that's not the only method erroring like that, there's a also the BleManagerGattCallback also within the GlucoseManager:

```
private final BleManagerGattCallback mGattCallback = new BleManagerGattCallback() {
  @Override
  protected Deque<Request> initGatt(final BluetoothGatt gatt) {
    ...
  }
  @Override
  public boolean isRequiredServiceSupported(final BluetoothGatt gatt) {
    ...
  }
  ...
  @Override
  protected void onCharacteristicWrite(final BluetoothGatt gatt, final BluetoothGattCharacteristic characteristic) {
    ...
  }
  ...
}

Every function within the BleManagerGattCallback which has @Override errors with "Method does not override method from its superclass".

Also with GlucoseManager, any method call using BleManagerCallbacks mCallbacks; states "Cannot resolve method 'methodNameHere(argumentHere)" such as:

```
mCallbacks.onDatasetChanged(gatt.getDevice());
...
mCallbacks.onNumberOfRecordsRequested(gatt.getDevice(), number);
...
mCallbacks.onOperationCompleted(gatt.getDevice());
...
mCallbacks.onOperationCompleted(mBluetoothDevice);
...
```

I'm not sure if I'm doing something wrong or misunderstanding, but there seems to be many issues with the GlucoseManager? I really appreciate you uploading this library, it has been a tremendous help and any further help would really be appreciated :).

philips77, 2018-06-15:
Hi, sorry for the delay.
I'll release an update to the nRF Toolbox today. There current version on master branch will be adjusted to match recent changes in the BLE library. Also, I'm working on version 2 of the BLE library (check [here](https://github.com/NordicSemiconductor/Android-BLE-Library/tree/develop). The nRF Toolbox is compatible with the new version also on `develop` branch. To compile the project, you also have to clone [BLE Common Lib](https://github.com/NordicSemiconductor/Android-BLE-Common-Library/tree/develop) . I'll try to release all libs on jcenter today/beginning of next week. 
Sorry for inconvenience.

Abushawish, 2018-06-15:
That's so awesome, thank you for the response :)!

philips77, 2018-06-19:
Hi, please pull latest changes. This should be fixed now.
Also, I'll move the current branch do version1 branch, and merger develop into master. Develop (and soon master) branch is using BLE Library v2.0-alpha1.

closed 2018-06-19 by philips77