nordicsemi/Android-nRF-Toolbox #54: Service does not fire onDeviceReady when previously connected

peabody124, 2018-08-29:
I was adapting my app (using the multiconnect profile) to keep the service persistent when switching between activities. One issue I ran into is that when the service is bound it fires an onDeviceConnected event if the device is connected. However at the level of how we interact with the devices it is much more pertinent to know if the onDeviceReady has fired since that is when all the characteristics have been discovered. It is also really quite awkward to know if the device has already fired that event as the is ```mInitInProgress``` variable does not have an accessor in the BleManager.

I'm happy to submit a pull request extending the ```BleMulticonnectProfileServiceReadyActivity``` and ```BleManager``` to track the ready state internally and when bound fire that event after firing onDeviceConnected. To me this seems like an elegant solution to the problem and does not require activities to attempt to infer the device connectivity state.

Thoughts?

philips77, 2019-01-30:
Hello,
Sorry for the super long delay. I'll look into your issue and try to come up with a solution in version 2.7.

philips77, 2019-04-02:
This will have to wait. We don't have time to fix this issue now. Most probably we will fix it together with rewriting the app in Kotlin. Unknown ETA. Let's leave it open.

philips77, 2019-04-02:
Feel free to submit a PR, I can merge and release a version quickly.

closed 2020-02-03 by philips77