nordicsemi/Android-nRF-Toolbox #57: Battery optimization is killing nrf-toolbox while running in the background

vineeshvs, 2018-10-29:
This happens in Android 7.+

Any workaround/ solutions?

Are you guys updating this app?

philips77, 2018-10-29:
Hi @vineeshvs,
Yes, nRF Toolbox is under maintenance.  Most new changes are on *develop* branch, but indeed I forgot about the battery optimization. The solution would be to keep running the service in foreground. Foreground services display a constant notification to the user, but the OS won't kill them automatically.
I'll fix the issue before the develop branch is merged to master, but it may take few days.
Thank you for creating the Issue.

philips77, 2019-04-02:
This will have to wait. We don't have time to fix this issue now. Most probably we will fix it together with rewriting the app in Kotlin. Unknown ETA. Let's leave it open.

archie94, 2023-04-12:
Does [this](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/main/lib_service/src/main/java/no/nordicsemi/android/service/NotificationService.kt) solve the issue described here?