nordicsemi/Android-nRF-Toolbox #58: Help Filtering Devices Out for UART

Casey10110, 2018-11-26:
Hi Guys,

Love this toolbox. Little issue though.

I have tried a lot to figure out how to filter out all the available devices for the UART activity. It seems very convoluted to me and I am having a difficult time figuring out how these devices even get listed. I've looked at DeviceListAdapter, DeviceAdapter, ScannerFragment. How could I, for example, only list devices that began with "My"?

Is this enough info what I am asking here?

Casey

philips77, 2018-11-26:
Hi @Casey10110,
Right now the toolbox's [ScannerFragment ](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/scanner/ScannerFragment.java) allows only to filter by service UUID, not by name. You'd have to modify the code here. You may either add a filter [here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/scanner/ScannerFragment.java#L244), or manually check the devices in [here](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/master/app/src/main/java/no/nordicsemi/android/nrftoolbox/scanner/ScannerFragment.java#L277).
Also, nRF Toolbox is quite old by now. It was written in a different age, so may not be up to date with the latest standards. You may check [nRF Blinky](https://github.com/NordicSemiconductor/Android-nRF-Blinky), which is much simpler, has a filter already implemented (service UUID or RSSI) and is much easier to modify. We use it for our apps if we need to start quickly.

Casey10110, 2018-11-26:
Excellent, thanks!

closed 2018-11-27 by philips77