nordicsemi/Android-nRF-Toolbox #59: Incorrect notification payloads delivered to app in specific circumstances

acutetech, 2018-11-28:
In a specific set of circumstances, the content of the Notifications delivered to an Android app contains the wrong data. The context is running the sample Glucose Application with nRF Toolbox: simulated glucose readings are delivered as a rapid sequence of notifications. The right number are delivered, but some are withheld from the app and are replaced by other notifications which were sent later. For example, when I expect to receive 20 notifications containing sequence numbers 0-19, I receive instead these notifications:

7, 7, 7, 15, 15, 15, 19, 19, 19, 19, 19, 19, 19, 19, 19, 19, 19, 19, 19, 19 

(the pattern of sequence numbers varies). 

When operating with the Nordic sample glucose app running on a PCA10040 periperal, and nRF Toolbox operating as the central, I see the problem with only one of my 4 devices (a Samsung Galaxy A5 phone) and only with the nRF Toolbox built from the Master release at Github. The problem is not there if I use nRF Toolbox downloaded from Playstore. I built Android-nRF-Toolbox-ble_lib-1.x from Github and that also does not show the fault.

To replicate:

Build the sample Glucose app described here:
http://infocenter.nordicsemi.com/topic/com.nordic.infocenter.sdk5.v15.2.0/ble_sdk_app_gls.html?cp=4_0_0_4_1_2_11

and install it on a PCA10040.

Install the nRF Toolbox on a Samsung Galaxy A5. Select the BGM app, then connect to the peripheral and bond to it (entering a 6-digit passkey displayed by the peripheral on a terminal). Press button 2 a number of times (say 10 times) to generate simulated readings. Press the "ALL" button on the phone to fetch the readings.

Toast pops up to say the expected number of records (e.g. 10) have been downloaded, but the "READINGS" panel will display fewer readings than expected (because many are duplicates). Viewing the nRF Logger log will show 10 notifications but examining the payloads reveals the fault reported.

I also see the fault with a sample Glucose app running on a CSR development system, and with a custom app that uses the BLE Library in the same way as nRF Toolbox. The problem is not seen if I add a large (c. 50ms) delay on the peripheral between sending each notification. I added instrumentation to the app (e.g. to pulse PIO pins on radio activity) and I cannot see differences in behaviour at that level.

I suspect the fault lies in the interaction between BLE Library (latest version) and the Samsung phone. The phone reports Model number SM-A520F, Android 8.0.0, Build number R16NW.A520FXXU7CRHA

The issue was raised previously here: https://devzone.nordicsemi.com/f/nordic-q-a/38515/frequency-of-notifications-with-android-ble-library

philips77, 2019-01-30:
Hi, this should be fixed in BLE Library 2.0.4. I forgot to release BLE Common Library 2.0.4, but I'll release 2.0.5 when BLE lib 2.0.5 is ready and then update the dependency in nRF Toolbox.

philips77, 2019-01-30:
This issue is blocked by: https://github.com/NordicSemiconductor/Android-BLE-Common-Library/issues/2

philips77, 2019-01-30:
Root cause of this issue: https://github.com/NordicSemiconductor/Android-BLE-Library/issues/6, fixed in 2.0.3.

philips77, 2019-04-02:
Fixed by #68.

closed 2019-04-02 by philips77