nordicsemi/Android-nRF-Toolbox #61: Android-nRF-Toolbox-2.6.0 and Android-BLE-Library-1.2.0

lzxsmile, 2018-11-30:
I use Android-BLE-Library-1.2.0 and Android-nRF-Toolbox-2.6.0,
EER: class Request not to find,why?
[image]
[image]

philips77, 2018-11-30:
Request was moved out of BleManager to a standalone class. Just add import as it will suggest.

lzxsmile, 2018-12-01:
i import the class,that method writeCharacteristic(mAlertLevelCharacteristic) 
[image]
will err.

lzxsmile, 2018-12-01:
[image]

lzxsmile, 2018-12-01:
import the first one
[image]
?

philips77, 2018-12-03:
Yes, the first one, as it suggests. Also, the data parameter has to be given as a parameter, not "inside" the `BluetoothGattCharacteristic` object.
That means, you should do:

```java
writeCharacteristic(mAlertLevelChar, someData);
```
instead of:
```java
mAlertLevelChar.setValue(someData);
writeCharacteristic(mAlertLevelChar);
```
This is to make sure the data you write will not override any data received in a notification, which may happen at the same time.

closed 2019-01-30 by philips77