nordicsemi/Android-nRF-Toolbox #63: Proguard error Android Studio 3.3

bluefirepatrick, 2019-01-31:
I have found the the proguard-rules.pro file throws an error in Android Studio 3.3  It seems to be because the pro guard file references a directory instead of a jar -which many be a requirement that changed somewhere along the way.

It's the first line of the file:
-libraryjars ../app/libs

I was able to fix it by adding the jar file to the line:
-libraryjars ../app/libs/achartengine-1.2.0.jar

which is the only jar in there.  I would he happy to submit a pull request if you would like me to.  I just thought I would mention it in case it helps.

Thanks for all the great Android code by the way.  I have been enjoying it.

philips77, 2019-01-31:
Hello! Please submit the PR. Thanks!

ananthrajsingh, 2019-02-28:
I also had the same issue.

philips77, 2019-04-02:
Fixed by #68

closed 2019-04-02 by philips77