nordicsemi/Android-nRF-Toolbox #65: Access is denied

kanubhav, 2019-02-12:
I downloaded the zip file and extracted in my desktop. I opened the file in AS and build it. When pressed the run button after some time it flashed me an error.
Error: C:\Users\H341261\Downloads\example\Android-nRF-Toolbox\app\..\app\libs, File not found: C:\Users\H341261\Downloads\example\Android-nRF-Toolbox\app\..\app\libs (Access is denied).
Can you tell me how to resolve this error?

philips77, 2019-02-12:
You should see this file in *libs* folder:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/tree/master/app/libs
Check Windows permissions for those folders. Perhaps it prevents from opening a jar file?
You may also clone the repo using git, not by downloading it in ZIP and try that way.
Sorry, but I can't help you more. It's related to your local setup on your computer.

kanubhav, 2019-02-12:
I tried to clone the repo using git but I am getting the same error.
In the folder: 
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/tree/master/app/libs
the .jar file is not opening for me.
I checked the windows permission also for that folder but still, I am not able to resolve the error.

wolfgang-elsner, 2019-02-25:
I have tried the same and have also the same error.

kanubhav, 2019-02-26:
Can you please provide a solution for the above issue!

thinkfeed, 2019-03-12:
maybe you shouldn't upgrade your com.android.tools.build:gradle version to latest

kanubhav, 2019-03-13:
Thanks @thinkfeed it worked.

closed 2019-03-13 by kanubhav

philips77, 2019-04-02:
Fixed by #68.