nordicsemi/Android-nRF-Toolbox #67: Notification channel on DFU on Android 9.0 needs to be updated

pedrochee, 2019-03-26:
Hi guys! The app is crashing on DFU, its due to the notificacion channel, I think the problem comes on NotificationActivity.java, I tried the following:

if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
			NotificationManager nm = (NotificationManager) getSystemService(NOTIFICATION_SERVICE);
			nm.createNotificationChannel(new NotificationChannel(NOTIFICATION_CHANNEL_ID_SERVICE, "App Service", NotificationManager.IMPORTANCE_DEFAULT));
			nm.createNotificationChannel(new NotificationChannel(NOTIFICATION_CHANNEL_ID_INFO, "Download Info", NotificationManager.IMPORTANCE_DEFAULT));
}

but its not working, any idea on how i can fix this?

philips77, 2019-03-27:
You may call a static method from the DfuServiceInitiator that will create the channel.
The app needs to be updated, we know. We just don't have resources right now.. :(

pedrochee, 2019-03-27:
Thanks @philips77 ill figure it out and post it here for future reference

pedrochee, 2019-03-28:
So I made it work, the solution is to: 

1.- add the foreground service permission in the manifest:

<uses-permission android:name="android.permission.FOREGROUND_SERVICE" />

2.- add static method on the OnCreate() of DfuInitiatorActivity.java to verify if the OS is running android O versions or above: (used the notification channel init on ToolboxApplication.java)

if (Build.VERSION.SDK_INT >= Build.VERSION_CODES.O) {
			DfuServiceInitiator.createDfuNotificationChannel(this);

			final NotificationChannel channel = new NotificationChannel(CONNECTED_DEVICE_CHANNEL, getString(R.string.channel_connected_devices_title), NotificationManager.IMPORTANCE_LOW);
			channel.setDescription(getString(R.string.channel_connected_devices_description));
			channel.setShowBadge(false);
			channel.setLockscreenVisibility(Notification.VISIBILITY_PUBLIC);

			final NotificationChannel fileChannel = new NotificationChannel(FILE_SAVED_CHANNEL, getString(R.string.channel_files_title), NotificationManager.IMPORTANCE_LOW);
			fileChannel.setDescription(getString(R.string.channel_files_description));
			fileChannel.setShowBadge(false);
			fileChannel.setLockscreenVisibility(Notification.VISIBILITY_PRIVATE);

			final NotificationChannel proximityChannel = new NotificationChannel(PROXIMITY_WARNINGS_CHANNEL, getString(R.string.channel_proximity_warnings_title), NotificationManager.IMPORTANCE_LOW);
			proximityChannel.setDescription(getString(R.string.channel_proximity_warnings_description));
			proximityChannel.setShowBadge(false);
			proximityChannel.setLockscreenVisibility(Notification.VISIBILITY_PUBLIC);

			final NotificationManager notificationManager = (NotificationManager) getSystemService(Context.NOTIFICATION_SERVICE);
			notificationManager.createNotificationChannel(channel);
			notificationManager.createNotificationChannel(fileChannel);
			notificationManager.createNotificationChannel(proximityChannel);
		}

This fixed the problem for me.

closed 2019-03-29 by philips77

hjh2881, 2020-06-03:
DfuServiceInitiator.createDfuNotificationChannel(context);

try this.