nordicsemi/Android-nRF-Toolbox #73: Error: Invoke-customs are only supported starting with Android O (--min-api 26)

romain145, 2019-05-06:
When trying to compile the project with Android Studio 3.4 this error is reported:

`Error: Invoke-customs are only supported starting with Android O (--min-api 26)`

The build.gradle file makes use of Java 8 as on the repo:

```
apply plugin: 'com.android.application'

android {
    compileSdkVersion 28
    buildToolsVersion "28.0.3"

    defaultConfig {
        applicationId "no.nordicsemi.android.nrftoolbox"
        minSdkVersion 18
        targetSdkVersion 28
        versionCode 69
        versionName "2.7.2"
        resConfigs "en"

        vectorDrawables.useSupportLibrary = true
    }
    buildTypes {
        debug {
            minifyEnabled true
            useProguard false
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
        release {
            minifyEnabled true
            useProguard true
            proguardFiles getDefaultProguardFile('proguard-android.txt'), 'proguard-rules.pro'
        }
    }
    compileOptions {
        targetCompatibility JavaVersion.VERSION_1_8
        sourceCompatibility JavaVersion.VERSION_1_8
    }
}

dependencies {
    implementation project(':common')
    implementation fileTree(dir: 'libs', include: ['*.jar'])
    wearApp project(path: ':wear')

    // nRF Toolbox is using Play Service 10.2.0 in order to make the app working in China:
    // https://developer.android.com/training/wearables/apps/creating-app-china.html
    //noinspection GradleDependency
    implementation 'com.google.android.gms:play-services-wearable:10.2.0'

    implementation 'androidx.appcompat:appcompat:1.1.0-alpha04'
    implementation 'androidx.preference:preference:1.1.0-alpha04'
    implementation 'com.google.android.material:material:1.1.0-alpha05'

    implementation 'no.nordicsemi.android:log:2.2.0'
    implementation 'no.nordicsemi.android.support.v18:scanner:1.4.0'

    // The DFU Library is imported automatically from jcenter:
    implementation 'no.nordicsemi.android:dfu:1.9.0'
    // if you desire to build the DFU Library, clone the
    // https://github.com/NordicSemiconductor/Android-DFU-Library project into DFULibrary folder,
    // add it as a module into the project structure and uncomment the following line
    // (and also the according lines in the settings.gradle):
    // implementation project(':dfu')

    // Import the BLE Common Library.
    // The BLE Common Library depends on BLE Library. It is enough to include the first one.
    implementation 'no.nordicsemi.android:ble-common:2.1.1'
    // The BLE Common Library may be included from jcenter. If you want to modify the code,
    // clone both projects from GitHub and replace the line above with the following
    // (and also the according lines in the settings.gradle):
    // implementation project(':ble-common')

    implementation('org.simpleframework:simple-xml:2.7.1') {
        exclude group: 'stax', module: 'stax-api'
        exclude group: 'xpp3', module: 'xpp3'
    }
}
```

maochiyu, 2019-06-12:
I had the same problem when  I trying to compile the project with Android Studio 3.4.1 . but I found  the(may not the latest version ) #41 may  help you solve the problem, [https://github.com/NordicSemiconductor/Android-nRF-Toolbox/issues/41#issuecomment-354993117](url),

maochiyu, 2019-06-12:
> When trying to compile the project with Android Studio 3.4 this error is reported:
>
> `Error: Invoke-customs are only supported starting with Android O (--min-api 26)`
>
> The build.gradle file makes use [...]

I had the same problem when I trying to compile the project with Android Studio 3.4.1 . but I found the(may not the latest version ) #41 may help you solve the problem, [https://github.com/NordicSemiconductor/Android-nRF-Toolbox/issues/41#issuecomment-354993117](url)

maochiyu, 2019-06-12:
[https://github.com/NordicSemiconductor/Android-nRF-Toolbox/issues/41#issuecomment-354993117]

maochiyu, 2019-06-12:
[https://github.com/NordicSemiconductor/Android-nRF-Toolbox/issues/41#issuecomment-354993117](url)

maochiyu, 2019-06-14:
> When trying to compile the project with Android Studio 3.4 this error is reported:
>
> `Error: Invoke-customs are only supported starting with Android O (--min-api 26)`
>
> The build.gradle file makes use [...]
try this，
Android Studio -> File -> Settings -> Build, Execution, Deployment -> Instant Run -> Disable checkbox