nordicsemi/Android-nRF-Toolbox #83: Couldn't find the  Bluetooth device

mnabilurrehman, 2019-11-18:
I downloaded this project and build it but the app wasn't able to find the my Bluetooth.

If I download your official app which is available on play store that app works fine in finding and connecting to the Bluetooth.

Can you please give me hint from where should I start find solution to this problem?

roshanrajaratnam, 2019-11-18:
Hi, have you granted the required permissions? If not you may go in to app settings and grant location permissions and also enable location services.

mnabilurrehman, 2019-11-18:
Yes! permission and service is granted. I have tested it on Samsung s10+ and huawei mate 10 lite. But the official play store app works fine.

philips77, 2019-11-18:
On Android 10 scanning requires LOCATION FINE PERMISSION. As far as I remember the app was requesting LOCATION COARSE PERMISSION, not fine. Go to Android Manifest and here:
https://github.com/NordicSemiconductor/Android-nRF-Toolbox/blob/08b67a28c9ab5c0dbc49be619c27209e2f955e30/app/src/main/java/no/nordicsemi/android/nrftoolbox/scanner/ScannerFragment.java#L228 and replace them with FINE version. Then try again.

mnabilurrehman, 2019-11-18:
The apps shows bonded bluetooths but not the one with nrf ble.

I have changed location to fine still it doesn't working.

philips77, 2019-11-18:
Perhaps you now have to Deny this permission and grant it again, so that the correct one is given?
The Play Store version targets API 28, so it may be in some legacy mode.