nordicsemi/Android-nRF-Toolbox #89: Error on writing characteristic (129)

cinny007, 2020-02-15:
I connected accu-chek Instant device to read glucose data. I pair the device and then click on ALL, a toast is shown   `Error on writing characteristic (129)`.

philips77, 2020-02-16:
Hello,
Does it happen every time you try? Are you able to send is logs from logcat?

YoungSooBae1224, 2020-02-17:
Hello,
I occured same issue. used nRF52810 and ble_app_gls example. IDE is SEGGER Embedded studio.
first, connect nRF52810 and Android device form nRF Toolbox.
second, store data(pushed button2) in nRF52810.
Third, I click on All button in nRF Toolbox BGM.
occured the error code 133(fatal error in nRF52810) - It did not occur in previous versions(2.7).
retry, changed error 129.

cinny007, 2020-02-17:
@philips77 Yes, It's happening always.

YoungSooBae1224, 2020-02-17:
[Log 2020-02-17_glucose.txt](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/files/4212091/Log.2020-02-17_glucose.txt)

I always happen too.
I don't know if it will help, Attach the log.

philips77, 2020-02-17:
What phone are you using? Did you try any other?

cinny007, 2020-02-17:
Redmi MI A3, Redmi note 4, Samsung M30....It's happening in all devices.

YoungSooBae1224, 2020-02-17:
I tried on SAMSUNG Galaxy Tab S3 Android 8.0.0 and SAMSUNG Galaxy Note4 Android 6.0.1.
Both phones are the same. The log is a Samsung Galaxy Tab S3.

After error 129, connecting to nRF Connect for Android, and Enable glucose measurement notifications and record access control point indications.
In this state, go to the nRF Toolbox and connect with the device. Pressing the All button will download the data but will not display anything on the screen.
Attach the log.
[Log 2020-02-17 174232.txt](https://github.com/NordicSemiconductor/Android-nRF-Toolbox/files/4212636/Log.2020-02-17.174232.txt)
This is also the same for both phones.

cinny007, 2020-02-18:
@philips77 Did you check?

philips77, 2020-02-20:
Hi,
I'm on it right now. Sorry, I was busy with other tasks. I managed to reproduce the issue, will investigate it now.

philips77, 2020-02-20:
Ok, I found the issue. The Glucose Measurement characteristic notifications were not enabled. Instead, the app was enabling Glucose Measurement Context characteristic notifications twice. The fix is on its way.

closed 2020-02-20 by philips77

cinny007, 2020-03-06:
Working fine now. Thanks @philips77 . I actually found one more issue for the same device. While inserting the strip in the accu-chek device, the application disconnects with the accu-chek device and hence I am not able to take the current reading. Shall I create one more issue for that?

philips77, 2020-03-06:
Yes, please create a new issue.